andrewrapp/xbee-arduino #2: Add to Arduino library manager

matthijskooijman, 2015-05-09:
Since 1.6.2, the Arduino IDE supports a library manager, to allow easily browsing and downloading libraries from the web. It would be convenient if this library was added. To do that:
 1. a library.properties file must be added to this repo
 2. a request must be sent to the Arduino devs

If you want, I can submit a PR for 1 and take care of 2?

andrewrapp, 2015-05-26:
I have to admit I haven't upgraded yet to the latest Arduino yet. Hopefully they solved the namespace problem where the library name (folder) must be unique. Yeah please PR

matthijskooijman, 2015-08-18:
@andrewrapp, Could you bump the library version number and create a git tag for it? The Arduino library manager needs a tag to include the library (and once it is included, any future tags will automatically cause the library to be updated in the library manager). AFAICS the tag does not need to have any particular format, I suspect the library manager looks at the library.properties file only.

matthijskooijman, 2015-09-09:
@andrewrapp, any chance you could take care of this (bumping version numbers and tagging)? Then I can get the library added to the Arduino library manager.

matthijskooijman, 2015-09-17:
Thanks for taking care of the tagging! I've filed the request with Arduino and this library is now available through the Arduino Library Manager.

closed 2015-09-17 by matthijskooijman