andrewrapp/xbee-arduino #10: Publish doxygen docs through gh-pages

matthijskooijman, 2015-08-02:
It seems doxygen docs are no longer available since github does not allow direct viewing of HTML files. However, it seems that by using gh-pages, you can again easily display HTML pages. This does mean that the HTML will be stored in a repository separate from the main repository, but that actually seems like a feature to me.

See this writeup for an example of how to set this up: http://rickfoosusa.blogspot.nl/2011/10/howto-use-doxygen-with-github.html

matthijskooijman, 2015-08-19:
Another attractive alternative could be to have readthedocs.org generate and publish the docs automatically. RTD is aimed at python documentation using sphinx, but [it can also be made to generate doxygen documentation](https://breathe.readthedocs.org/en/latest/readthedocs.html) and also add separate overview documentation using the powerful rst format as well.

andrewrapp, 2015-11-18:
It seems that doxygen now supports markdown https://www.stack.nl/~dimitri/doxygen/manual/markdown.html