andrewrapp/xbee-arduino #13: SoftwareSerial not working with XBee

ClintonKeith, 2015-10-13:
I have the 1.6.5 IDE and the latest (10/6) XBee library.  When I try to connect my XBee with a SoftSerial interface, the Uno is resetting.

I use the following in Series2_Tx setup:
  SoftwareSerial mySerial(8,9); // RX, TX
  mySerial.begin(9600);
  xbee.setSerial(mySerial);   

and as soon as xbee.send(zbTx); is called the Uno reboots.

davidsainty, 2015-10-13:
Looks like mySerial is a local variable in setup ()?

Make it a global.

>
> I have the 1.6.5 IDE and the latest (10/6) XBee library. When I try to
connect my XBee with a SoftSerial interface, the Uno is resetting.

matthijskooijman, 2015-10-13:
@davidsainty seems to point out the cause of your error.

Additionally, you might want to consider using the AltSoftSerial library, which is a lot more reliable and can do full-duplex communication as well (e.g. receive while transmitting), which SoftwareSerial doesn't. It's limitation is that it only works on pin 8 and 9 (on a Uno), but it seems you're already using those anyway.

ClintonKeith, 2015-10-13:
Ugh.  Well I guess I picked the wrong week to start sniffing glue.

Thanks for catching my blunder and the advice on AltSoftSerial.

closed 2015-10-13 by ClintonKeith