andrewrapp/xbee-arduino #19: Support for devices without hardware Serial

julesra, 2015-12-06:
I tried running this on a Trinket (which runs ATtiny chip), and that device doesn't have a hardware `Serial`, therefore the XBee.cpp file doesn't compile because `XBee::XBee(): _response(XBeeResponse())` tries to do: `_serial = &Serial;`

The way I fixed my compilation problem is by editing the `_response` function and setting `_serial = 0;` but that seems like a hack. (PS: I ran into some other problems with my Trinkets, so I don't know if that actually worked).

pepib, 2015-12-12:
I've the same problem when I try connect an Xbee Series 2 to Arduino Due in API mode. In fact I'm able to receive a package (with "Hello" string) on my computer sended by arduino. But when I try to send a trasmit request 16-bit address (with "Hello" string) from my computer to arduino getResponse(), I think, that is always null

julesra, 2015-12-14:
@pepib: I don't think that is related to my issue. Does your Arduino code compile on Due ? If yes, then your problem is not related to the issue I reported.

Sleepnow, 2016-01-31:
It is a hack, but it's the only reference to Serial in the code.  Looks like it should work?  Will dig out a trinket and test it later.