andrewrapp/xbee-arduino #20: Mega Rx from Xbee causing high voltage 1023 on all analog input pins

profitablecrop, 2015-12-21:
Hello,

Having a very strange issue that seems associated to the Tx packet from the Xbee to the Mega Rx (on any serial port) which then causes a stuck high voltage on all the mega input pins.  Nothing seems to reset the high voltage (1023 analog reading on any input pin) except a hard reset which then reads normal until a the xbee is attached.  Prior to updating the latest Xbee lib, it would only start the issue when a Rx packet came through to the mega, now it happens when I connect the Tx from the Xbee to the Rx of the mega pretty quickly. and hangs as it had before.  Not sure if you might understand more of how the Rx packets might affect the analog input readings of the mega.

Thanks.

Alec

profitablecrop, 2015-12-22:
Problem actually appears code related somehow after I tried to scrap away code..also the hardware reasoning would not make much sense.  I gutted some blocks of code and issue went away... will post back with more info if I find the cause for why the analog pin readings were being affected.. For whatever reason the Rx serial data from xbee was causing an issue with the analog input voltage readings, which would remain high at 1023 even if you then took the pin and grounded it while looping..readings would not zero out.

profitablecrop, 2015-12-23:
Backing up and looking for more info on the analog pins, I found that there must have been an issue with the reference voltage.
I adjusted the mega reference voltage to External and fed 5V to the pin, seems to have solved the issue.  I am using the Arduino GSM shield and Xbee on the Mega board, these 2 items seem to be affecting the AREF somehow.

matthijskooijman, 2015-12-23:
What XBee shield are you using? Did you try with just the XBee shield or just the GSM shield?

ClintonKeith, 2015-12-23:
Are you just jumping the 5V pin to aRef?  If so, I'd suspect that the voltage regulator is struggling a bit here.  Are you just plugged into USB?

profitablecrop, 2015-12-23:
Direct Xbee or an Xbee adaptor directly to the board, 5v/3.3v, GND, Tx, Rx
pins only, no shield for the xbee.  Using the GSM shield on the Mega.
If I took off the xbee worked fine, and if I omitted some GSM begin code it
worked fine,
but both together caused some sort of interference.
The 5V direct to the AREF pin seemed to fix the issue.

*Best Regards,*

*Alec Mayall*
Horticulturalist, CCA, PCA, CAIS
Profitable Crop Productions LLC
951-847-2131
profitablecrop.com

ClintonKeith, 2015-12-23:
But what is supplying the power to the Arduino?  Some USB hubs don't supply enough current and might show what you are seeing.  The VCC that you're plugging directly into the aref (with EXTERNAL) acts the same as using the INTERNAL flag, except that the 5V that is used for the internal reference is further from the voltage regulator.   

While your solution works, you might be seeing some other low-current effects elsewhere.  I'd test this by plugging in a 9V power supply.

profitablecrop, 2015-12-26:
I have a custom board built more or less as a shield on the mega, and then
the GSM shield on my shield.  Powered from a 12v SLA battery with
regulators for the mega @ 7.5V, GSM @ 5V, and XBee @ 3.3V.

Not sure why adjusting to EXTERNAL and providing the 5V directly to the
AREF pin worked, but that is the only thing that allowed me to get real
analog readings.  I thought it may have been related to my shield, but had
the same issue with just the Arduino GSM shield on the Mega where the 5v to
AREF jumper solved it.

The problem began, or I noticed it rather, when I was trying to read the
main units battery voltage on the 12V via a voltage divider and an analog
input pin.  I kept getting an analog input of 1023 after some interference
somewhere, which I knew that wasn't correct, so I began trying to see why I
could not get accurate analog pin readings.

Things are looking fine now.

*Best Regards,*

*Alec Mayall*
Horticulturalist, CCA, PCA, CAIS
Profitable Crop Productions LLC
951-847-2131
profitablecrop.com