andrewrapp/xbee-arduino #24: Can't get two xbee's to communicate...

happilymarrieddad, 2016-03-25:
Hey guys, I can't seem to get these two xbee's to communicate. I really want to get this project finished so I can finally fly my drone!!! Any help would be appreciated. Thanks!

Here is my xbee testing code.

[image]

[image]

[image]

[image]

davidsainty, 2016-03-25:
Normally you would configure one xbee as a coordinator, one as a router,
and probably set up encryption keys as well. Have you done all of that
first?

happilymarrieddad, 2016-03-25:
No, I got it to work once just using the github.com library and broadcasting using the unique id of each xbee... now it's been a while and I can't remember how I did it lol. I never did any of that and I was still able to receive data.

davidsainty, 2016-03-25:
Did you use these same two S3B XBees, or some other XBees?

happilymarrieddad, 2016-03-25:
No, but the same generation just different ID's.. I still have the ones that worked and I tried them first but they weren't working so I ordered two more of the same, thinking maybe they were damaged by ESD or something and the new ones aren't receiving a signal either.

paolochiarlone, 2016-03-25:
I would not setup the Encryption, that adds more complication and you do not need it.
The XBee that is setup as Coordinator needs to have the address of the other node (64 bit).
The Router node does not, as it will reply back to the Coordinator by default, but it needs 0x00000000 as a destination address
Make sure they are configured correctly.
Also, make sure they are joined in the same network, as the Router might create his own network if it does not get a join ACK from the coordinator (i.e. you turned on the Router first).
I hope this helps.

happilymarrieddad, 2016-03-30:
I think there is a problem here... these are RF modules and they use 900MHZ to communicate. They aren't the WIFI modules. These are the 28 mile Series Pro 2 modules.

paolochiarlone, 2016-03-30:
@happilymarrieddad 
The Series2 (PRO or non PRO) is NOT 900 MHz, it is 2.4 GHz ZigBee which WILL NOT communicate via WiFi.
If you need WiFi, you need to look into Series6 (XBee WiFi).
If you refer to 900 MHz 28 miles, then you're talking about Series 3.
This might help:
http://www.digi.com/pdf/chart_xbee_rf_features.pdf

davidsainty, 2016-03-30:
Actually they are series 3 in the images.

Which might mean my coordinator/router setup "help" doesn't apply at all -
I'd hastily concluded you were using zigbee, which I guess you are not.

happilymarrieddad, 2016-03-31:
Hey guys thanks so much for the time you've taking to help me. I've verified that it is the XBEE-PRO® 900HP. Do you guys know of any tutorials to set it up? Thanks!

paolochiarlone, 2016-03-31:
I would read the documentation here:
http://www.digi.com/products/xbee-rf-solutions/modules/xbee-pro-900hp#productsupport