andrewrapp/xbee-arduino #27: Send issue

PreciousRoy0, 2016-04-07:
Hello,

i have been spending the last day or so trying to get my 2 (soon to be 3) xbee pro S2B to communicate with each other. in AT mode that works fine the both can send and receive.

when in API mode the Coordinator can send message to my router node and it receives just fine, but when running the TX sample code it keeps on failing. 

so i decided to edit the receive code to echo back anything it gets. but coordinator does not receive anything back. do any of you know what could be my problem?

```
#include <XBee.h>
#include <SoftwareSerial.h>

/*
This example is for Series 2 XBee
Sends a ZB TX request with the value of analogRead(pin5) and checks the status response for success
*/

// create the XBee object
XBee xbee = XBee();

uint8_t payload[] = { 'H', 'i' };

// SH + SL Address of receiving XBee //0013A200 40B7B2B8
XBeeAddress64 addr64 = XBeeAddress64(0x0013a200, 0x40B7B2B8);
ZBTxRequest zbTx = ZBTxRequest(addr64, payload, sizeof(payload));
ZBTxStatusResponse txStatus = ZBTxStatusResponse();

// create reusable response objects for responses we expect to handle 
ZBRxResponse rx = ZBRxResponse();
ModemStatusResponse msr = ModemStatusResponse();

// Define NewSoftSerial TX/RX pins
// Connect Arduino pin 8 to TX of usb-serial device
uint8_t ssRX = 9;
// Connect Arduino pin 9 to RX of usb-serial device
uint8_t ssTX = 8;
// Remember to connect all devices to a common Ground: XBee, Arduino and USB-Serial device
SoftwareSerial nss(ssRX, ssTX);

void setup() {
	// start serial
	Serial.begin(9600);
	nss.begin(9600);

	xbee.setSerial(nss);

	Serial.println("Starting up!");
}

void loop() 
{
	xbee.readPacket();

	if (xbee.getResponse().isAvailable())
	{
		// now fill our zb rx class
		xbee.getResponse().getZBRxResponse(rx);

		// got a response!
		Serial.println("got a response!");

		Serial.println("sending back.");
		zbTx = ZBTxRequest(rx.getRemoteAddress64(), rx.getFrameData(), sizeof(rx.getFrameData()));

		xbee.send(zbTx);

	}
	else if (xbee.getResponse().isError())
	{
		Serial.print("Error reading packet.  Error code: ");  
		Serial.println(xbee.getResponse().getErrorCode());
	}
}
```
The settings for the xbee's are:
Coordinator:
```
<?xml version="1.0" encoding="UTF-8"?>
<data>
  <profile>
    <description_file>XBP24-ZB_21A7_S2B.xml</description_file>
    <settings>
      <setting command="ID">3613</setting>
      <setting command="SC">000C</setting>
      <setting command="SD">3</setting>
      <setting command="ZS">0</setting>
      <setting command="NJ">FF</setting>
      <setting command="DH">0</setting>
      <setting command="DL">FFFF</setting>
      <setting command="NI">0x20</setting>
      <setting command="NH">1E</setting>
      <setting command="BH">0</setting>
      <setting command="AR">FF</setting>
      <setting command="DD">30000</setting>
      <setting command="NT">3C</setting>
      <setting command="NO">0</setting>
      <setting command="CR">3</setting>
      <setting command="PL">4</setting>
      <setting command="PM">1</setting>
      <setting command="EE">0</setting>
      <setting command="EO">0</setting>
      <setting command="KY"></setting>
      <setting command="NK"></setting>
      <setting command="BD">3</setting>
      <setting command="NB">0</setting>
      <setting command="SB">0</setting>
      <setting command="D7">1</setting>
      <setting command="D6">0</setting>
      <setting command="AP">1</setting>
      <setting command="AO">0</setting>
      <setting command="SP">20</setting>
      <setting command="SN">1</setting>
      <setting command="D0">1</setting>
      <setting command="D1">0</setting>
      <setting command="D2">0</setting>
      <setting command="D3">0</setting>
      <setting command="D4">0</setting>
      <setting command="D5">1</setting>
      <setting command="P0">1</setting>
      <setting command="P1">0</setting>
      <setting command="P2">0</setting>
      <setting command="PR">1FFF</setting>
      <setting command="LT">0</setting>
      <setting command="RP">28</setting>
      <setting command="DO">1</setting>
      <setting command="IR">0</setting>
      <setting command="IC">0</setting>
      <setting command="V+">0</setting>
    </settings>
  </profile>
</data>
```
and node:
```
<?xml version="1.0" encoding="UTF-8"?>
<data>
  <profile>
    <description_file>XBP24-ZB_23A7_S2B.xml</description_file>
    <settings>
      <setting command="ID">3613</setting>
      <setting command="SC">C</setting>
      <setting command="SD">3</setting>
      <setting command="ZS">0</setting>
      <setting command="NJ">FF</setting>
      <setting command="NW">0</setting>
      <setting command="JV">0</setting>
      <setting command="JN">0</setting>
      <setting command="DH">0013A200</setting>
      <setting command="DL">40B7B2B8</setting>
      <setting command="NI">0x20</setting>
      <setting command="NH">1E</setting>
      <setting command="BH">0</setting>
      <setting command="AR">FF</setting>
      <setting command="DD">30000</setting>
      <setting command="NT">3C</setting>
      <setting command="NO">0</setting>
      <setting command="CR">3</setting>
      <setting command="PL">4</setting>
      <setting command="PM">1</setting>
      <setting command="EE">0</setting>
      <setting command="EO">0</setting>
      <setting command="KY"></setting>
      <setting command="BD">3</setting>
      <setting command="NB">0</setting>
      <setting command="SB">0</setting>
      <setting command="D7">1</setting>
      <setting command="D6">0</setting>
      <setting command="AP">1</setting>
      <setting command="AO">0</setting>
      <setting command="SM">0</setting>
      <setting command="SN">1</setting>
      <setting command="SO">0</setting>
      <setting command="SP">20</setting>
      <setting command="ST">1388</setting>
      <setting command="PO">0</setting>
      <setting command="D0">1</setting>
      <setting command="D1">0</setting>
      <setting command="D2">0</setting>
      <setting command="D3">0</setting>
      <setting command="D4">0</setting>
      <setting command="D5">1</setting>
      <setting command="P0">1</setting>
      <setting command="P1">0</setting>
      <setting command="P2">0</setting>
      <setting command="PR">1FFF</setting>
      <setting command="LT">0</setting>
      <setting command="RP">28</setting>
      <setting command="DO">1</setting>
      <setting command="IR">0</setting>
      <setting command="IC">0</setting>
      <setting command="V+">0</setting>
    </settings>
  </profile>
</data>
```

davidsainty, 2016-04-08:
Just some things to try:

> SoftwareSerial nss(ssRX, ssTX);

SoftwareSerial is half-duplex, which can cause problems.  Hardware is best,
and AltSoftSerial also works well.

> zbTx = ZBTxRequest(rx.getRemoteAddress64(), rx.getFrameData(),
sizeof(rx.getFrameData()));

That sizeof() is definitely not right :)  Maybe rx.getFrameDataLength()
instead?

PreciousRoy0, 2016-04-08:
about the serial connection... The uno only has one serial port and i need a way to debug. 

so i changed the cod to this as per your suggestion. where instead of sending the data back i just send the word hi. 

`zbTx = ZBTxRequest(rx.getRemoteAddress64(), payload, sizeof(payload)); `

```
#include <XBee.h>
#include <AltSoftSerial.h>

// create the XBee object
XBee xbee = XBee();

uint8_t payload[] = { 'H', 'i' };

// SH + SL Address of receiving XBee //0013A200 40B7B2B8
ZBTxRequest zbTx;

// create reusable response objects for responses we expect to handle 
ZBRxResponse rx = ZBRxResponse();
ModemStatusResponse msr = ModemStatusResponse();

AltSoftSerial nss;

uint8_t tmp[] = { 0x7E, 0x00, 0x14, 0x10, 0x01, 0x00, 0x13, 0xA2, 0x00, 0x40, 0xB7, 0xB2, 0xB8, 0xFF, 0xFE, 0x00, 0x00, 0x62, 0x61, 0x6E, 0x61, 0x6E, 0x61, 0x7A }; //manual frame works

void setup() {
	// start serial
	Serial.begin(9600);

	nss.begin(9600);
	xbee.setSerial(nss);

	Serial.println("Starting up!");

	for (size_t i = 0; i < sizeof(tmp); i++) // this works 
	{
		nss.write(tmp[i]);
		Serial.print(tmp[i], HEX);
		Serial.print(",");
	}
}

void loop() 
{
	xbee.readPacket();

	if (xbee.getResponse().isAvailable())
	{
		// now fill our zb rx class
		xbee.getResponse().getZBRxResponse(rx);

		// got a response!
		Serial.println("got a response!");

		Serial.println("sending back.");
		zbTx = ZBTxRequest(rx.getRemoteAddress64(), payload, sizeof(payload)); 

		xbee.send(zbTx);//this fails
	}
	else if (xbee.getResponse().isError())
	{
		Serial.print("Error reading packet.  Error code: ");  
		Serial.println(xbee.getResponse().getErrorCode());
	}
}
```

But i am still not receiving the data so i will do some more test and get back to you.

PreciousRoy0, 2016-04-08:
Ok after some more debugging, when i edit the sendByte function and only send the byte data the code works just fine. and i receive the data on the other side.

```
void XBee::sendByte(uint8_t b, bool escape) 
{
	//Serial.print(b, HEX);
	if (escape && (b == START_BYTE || b == ESCAPE || b == XON || b == XOFF)) 
	{
		//write(ESCAPE);
		//write(b ^ 0x20);

		write(b);

	} else {
		write(b);
	}
}
```

can someone tell me what exactly those lines of code do? and why my setup is wrong/not working?

davidsainty, 2016-04-08:
Ah.  Good work.

You need AP to be 2 in the Xbee configuration.

That's what that line is implementing.

You still need to change sizeof () in your code.

PreciousRoy0, 2016-04-10:
Thanks  davidsainty: that did indeed fix my problem. i could not find it so it might not be mentioned but it could be a good idea to have the settings for the xbee mentioned somewhere that would of saved me a days work. thanks again for the help ;}

matthijskooijman, 2016-04-27:
@PreciousRoy0, you mean something like here, under "Configuration"? https://github.com/andrewrapp/xbee-arduino#configuration

jortronm, 2016-06-05:
Hi,
Even though I have not been able to successfully use the library yet I would like to thanks Andrew Rapp for the library and all the people supporting this project. Great job.
I have spent a great deal of time trying to replicate the series 2 rx example but it doesn’t matter what I do the Arduino keeps on generating the wrong API message.

I’m using the following software to send a “Hi” message.
The message should look like 7E 00 7D 33 10 01 00 7D 33 A2 00 40 DA 9D 23 FF FE 00 00 48 65 6C 6C 6F 6E but the Arduino is generating this 7E 01 21 01 07 D3 3A 20 40 DA 9D 23 FF FE 00 48 65 6C 6C DD which is not even a valid API message

I will really appreciate some help or guidance on what I might be doing wrong.

Thanks
George

#include <XBee.h>
#include <SoftwareSerial.h>

// Create an XBee object at the top of your sketch
XBee xbee = XBee();

int xBeeRX = 6; // Connect Arduino pin 8 to Xbee TX pin
int xBeeTX = 7; // Connect Arduino pin 9 to Xbee RX pin
SoftwareSerial xBeeSerial(xBeeRX, xBeeTX);// Define SoftwareSerial xBeeSerial RX/TX pins

// Create an array for holding the data you want to send.
uint8_t payload[] = {'H','e','l','l'}; //{0x48,0x65,0x6C,0X6C,0X6F} in HEX

// Specify the address of the remote XBee (this is the SH + SL)
XBeeAddress64 addr64 = XBeeAddress64(0x0013A200, 0x40DA9D23);//0013A20040DA9D23

// Create a TX Request
//The API msg for this should be   7E 00 7D 33 10 01 00 7D 33 A2 00 40 DA 9D 23 FF FE 00 00 48 65 6C 6C 6F 6E
//but this is what Im getting 7E 01 21 01 07 D3 3A 20 40 DA 9D 23 FF FE 00 48 65 6C 6C DD
ZBTxRequest zbTx = ZBTxRequest(addr64, payload, sizeof(payload));

void setup() {
  Serial.begin(9600); // Start the xBeeSerial port
  xbee.setSerial(Serial); // Tell XBee to use  SoftwareSerial xBeeSerial
}

void loop() {
xbee.send(zbTx);// Send your request
  delay(5000);
}

davidsainty, 2016-06-06:
7E 00 7D 33 10 01 00 7D 33 A2 00 40 DA 9D 23 FF FE 00 00 48 65 6C 6C 6F 6E
7E 01 21 01         07 D3 3A 20  40 DA 9D 23 FF FE 00    48 65 6C 6C DD

I get the impression you've got some kind of serial-level framing issue
going on.  In hex, it sure looks like it's sometimes getting four bits out
of sync.  And zero bytes seem to cause particular difficulty.

What baud rate are you using?  And is the chip using a crystal, or is it
using the internal clock?

jortronm, 2016-06-06:
Hi David,
Thanks for getting back to me.
Yes, it seems like some of the bits are there but others are missing.
I have tried different baud rates 300, 2400 & 9600 and I’m still get the same wrong API frame.
I have tried the code in UNO ATmMega328 and Nano ATmega 328 and the result is the same.
I’m using the internal clock of the Arduinos.
What else should I try?

davidsainty, 2016-06-06:
I think the internal clock is not precise enough for serial timing. I
worked with someone else trying to use the internal clock for serial, and
whilst I can't recall what the maths came out as, I do recall the
conclusion was to use a crystal.

You could try calibrating the internal clock, but if you want to prove it
works first, try using a crystal.

jortronm, 2016-06-06:
I will give the crystal a go. thanks

Suraj151, 2016-10-20:
Hello friends, new here..faced same issue...but solved..by some modifications...well i didn't get what is XON  
& XOFF for..??
[image]

and it used here
[image]

which add escape if ur XBee address contain 0x11 or 0x13 value. need to resolve. for now i commented them due to same error...its work fine..but reply for what XON & XOFF r there..??

PreciousRoy0, 2016-10-20:
Suraj151 i have no idea what thy do exactly, i image it accounts for some kind of mode.

can you post in code! your basic test code so we can have a look how you set it all up.

Suraj151, 2016-10-20:
sure PreciousRoy0 as i told i just commented XON and XOFF for now...as in below snap

[image]

[image]

its simple haha...until u don't know what it used for...we can skip if not required...well also some simplicity purpose issue i have posted....we can make it more simple to use...

PreciousRoy0, 2016-10-20:
no i mean look at the first post in this thread, i give the most basic implementation of failing code, your code not lib code that shows how you set it up, and your xbee settings can also be helpful

Suraj151, 2016-10-20:
Hello evening friend,
PreciousRoy0 i just checked ur coordinator and end device setting it seems like same as mine...as my end is XBee S2 series 2 module and ur XBee Pro S2B so some parameter are not in my setting as in ur side.

well i will share my both end code....its not Echo type..mean we can made it by program...i m just sending sample text from coordinator (arduino) and i get it on end device (PC).
Coordinator code:

#include <XBee.h>
#include <SoftwareSerial.h>

uint8_t ssRX = 8;
uint8_t ssTX = 9;
SoftwareSerial SoftSerial(ssRX, ssTX);

XBee xbee = XBee();

char payload[] = "Hello End Device";

XBeeAddress64 addr64 = XBeeAddress64(0x0013A200, 0x41241CB2);

ZBTxRequest Tx = ZBTxRequest(addr64, payload, sizeof(payload));
ZBTxStatusResponse Txresponse = ZBTxStatusResponse();
char temp[5];

void setup() { 
  Serial.begin(9600);
  xbee.setSerial(SoftSerial);
  SoftSerial.begin( 9600 );
  delay(3000);
}

void loop() {
 xbee.send(Tx);
 delay(500);
  xbee.readPacket();

  if (xbee.getResponse().isAvailable()) {

    if (xbee.getResponse().getApiId() == ZB_TX_STATUS_RESPONSE) {
      xbee.getResponse().getZBTxStatusResponse(Txresponse);
      Serial.print("Delivery status : ");
      Serial.println(Txresponse.getDeliveryStatus(),HEX);
    } 
    else {
      Serial.print("Expected TX Response, but got Frame : ");
      Serial.print(xbee.getResponse().getApiId(), HEX);
      Serial.println();
    }    
  } else if (xbee.getResponse().isError()) {
    Serial.print("Error reading packet.  Error code: ");  
    Serial.println(xbee.getResponse().getErrorCode());
  }
  delay(3000);
}

on PC terminal we can see received msg....

PreciousRoy0, 2016-10-20:
it seems to look ok, i would make some changes though.
use AltSoftSerial i had som dificultys with SoftwareSerial =https://www.pjrc.com/teensy/td_libs_AltSoftSerial.html
second i would open the serial connection before you pass it on to the xbee 
other then that to me it looks fine, all i can think of is that it is a setting of your xbee and i do not know enough to comment on that, you will need  davidsainty for that

Suraj151, 2016-10-20:
well, no matter i have done enough example with this library....sure we can make it best
anybody know about why we need that callback function in xbee lib..i mean i want to use them...but wanna know for what purpose we need it.