andrewrapp/xbee-arduino #28: Explicit Addressing ZigBee Command frame support

Dariostesh, 2016-05-02:
Dear Andrew,

does your Arduino - XBee API mode library support the use of the Explicit Addressing ZigBee Command frame?

Thanks in advance,

Regards,

Dario

matthijskooijman, 2016-05-02:
Yup, here: https://github.com/andrewrapp/xbee-arduino/blob/master/XBee.h#L1226

And see https://github.com/andrewrapp/xbee-arduino/blob/master/examples/ZdpScan/ZdpScan.ino for an example that uses it.

Dariostesh, 2016-05-04:
Ok, thanks, so do I just need to look at these two files? Initially, I have seen the XBee.h and there is written that not all the API frame are supported. However, I have read something about the ZDP scan, but I think I need a little bit more time to investigate. Since now, I have used the XCTU software Digi provide to users, and in there I was able to create easily an API frame putting the parameters (endpoints, cluster ID, profile ID) in some fields, finally sending the frame to the EUI64 destination. The ZdpScan.ino sketch is quite long, and I want to ask if its use is as simple as putting the parameters in some pre-established fields, and then loading the sketch inside Arduino.

Thanks,

Regards,

Dario

Dariostesh, 2016-05-06:
Matthijs,

I have investigated on the files. So, do your code support the specification of different profile ID like 0xC091 and/or different cluster ID and/or different endpoints rather than the ones included in the file zigbee.h inside the ZDPScan folder example? I mean, my objective is to send a TX explicit addressing command message with these parameters:

- Source Endpoint: E8 (XBee Digi endpoint used)
- Destination Endpoint: 01 (ETRX357 Telegesis endpoint used for their command set)
- Cluster ID: 0x0002 (Telegesis cluster ID used)
- Profile ID: 0xC091 (Telegesis Profile ID)
- Payload: "3,A,500,0" (without quotes)

and it seems that it requires additional fields to send this kind of TX packet with the API mode observing the ZigBee.h file even though the XBee.h file contains the TX Explicit command. My question is:
1) Do I need to manually add (#define) to the ZigBee.h file the destination endpoints, cluster ID and Profile ID to which I want to send the frames packet? 2) In the XBee.h file there is the TX Explicit Command where you can specify the above cited parameters. In which form the parameters (endpoints, cluster ID and Profile ID) have to be declared inside a normal Arduino sketch?

Thanks in advance for your replying,

Kind Regards,

Dario

Dariostesh, 2016-05-09:
Hello Matthijs,

I am using this part of code to send an explicit TX message:

#include <XBee.h>

// Create an XBee object at the top of your sketch
XBee xbee = XBee();

void setup() {
  // put your setup code here, to run once:

  // Start the serial port
Serial.begin(9600);
// Tell XBee to use Hardware Serial. It's also possible to use SoftwareSerial
xbee.setSerial(Serial);

}

void loop() {
  // put your main code here, to run repeatedly:

  // Create an array for holding the data you want to send.
uint8_t payload[] = { '3', ',', 'A', ',', '5', '0', '0', ',', '0' };

// Specify the address of the remote XBee (this is the SH + SL)
XBeeAddress64 addr64 = XBeeAddress64(0x000D6F00, 0x0AB9FA93);

  // Create an explicit TX Request
ZBTxRequest zbTx = ZBExplicitTxRequest(addr64, 0xfffe, 0, 0, payload, sizeof(payload), 0, 0xE8, 0x01, 0x0002, 0xC091);

// Send your request
xbee.send(zbTx);

delay(2000);

}

I'm using Arduino One, but when connecting the XBee shield and the XBee too to it, nothing seems to be transmitted. The jumpers are on the XBee position. Do you know why this is happening?

Thanks,

Kind Regards,

Dario

paolochiarlone, 2016-05-09:
@Dariostesh 
Have you checked the 64-bit address?
0x000D6F00, 0x0AB9FA93 is non-Digi address.
If it were a Digi XBee it should start with 0x0013A200 40.......
Paolo

Dariostesh, 2016-05-10:
@paolochiarlone 

I will continue talk in english to ensure everyone can read what we write.
So, Thanks for commenting. The address is relative to a Telegesis radio module, in particular the ETRX357 module, which has that starting alphanumeric MAC address. The purpose is properly to send a frame packet to a specific radio module in order to turn the lights on on my electronic board. I have already tried to send the frame packet with XCTU software of Digi and it works well using the explicit addressing command frame (API frame ID 0x11) with the parameters reported above, as Silicon Labs suggested me.
Right now, the problem is to send this specific command with Arduino One, and it seems possible with this library, but when I connect the XBee shield with XBee itself to Arduino, the TX led on it stop and it seems that nothing is transmitted to XBee, even though I load the sketch on Arduino One. Without the shield on it, Arduino transmits regularly to the serial monitor (uncomprehensible messages apart from the payload). Am I wrong in doing something? I have already put the jumpers in the XBee position.

Thanks,

Regards,

Dario

Dariostesh, 2016-05-13:
Ok, I want to narrow the problem down. Basically I think the code is not a problem, even though it can be not perfectly.
I have now bought an Arduino Leonardo and uploaded the code using both the serial ports, one for debugging on the computer and one for sending frame packets to XBee. However, the real problem is that when I put the XBee shield and the XBee itself on Arduino Leonardo (or Arduino One also), this seems not to transmit anything since the 13 LED become turned on permanently like Arduino it's waiting for a bootloading (I have read of this on the Arduino forum) and the TX led never turn on to show that something has been transmitted. Could it be a serial hardware problem? I have investigated on the argument and found that Arduino probably receive garbage from XBee so thinking that something is being uploaded to it but not really happening. Could it be the reason of this Arduino stucking?

Thanks,

Regards,

Dario P.

paolochiarlone, 2016-05-14:
@Dariostesh 
Unfortunately I have no experience in Telegesis products, so I cannot really help on this, sorry :(