andrewrapp/xbee-arduino #33: error: invalid conversion from 'uint8_t* {aka unsigned char*}' to 'char*'

Aragantinis, 2016-10-20:
Its giving this error when trying to complile.

I think its a recent change.

I'm compiling using platformiIO.

Full error:
lib/xbee-arduino-master/XBee.cpp: In member function 'char* NodeIdentifierResponse::getNodeIdentifierString()':
lib/xbee-arduino-master/XBee.cpp:787:27: error: invalid conversion from 'uint8_t* {aka unsigned char*}' to 'char*' [-fpermissive]
return getFrameData() + 21;

^
*** [.pioenvs/megaatmega2560/lib/xbee-arduino-master/XBee.o] Error 1

sodcorec, 2016-11-01:
Has anyone found the anwser to this problem. I am getting the same thing on Android IDE linux.
 Please help or investigate.
 Thank you,
 Donald

sodcorec, 2016-11-01:
In the cpp file and the header file I changed the return value of char to uint8_t.
 There does not seem to be anything to be calling that function at present from the header or .cpp file, so it will work in the short them.
 I am not sure what they are all trying to do, so if you call the function getNodeIdentifierString() it may not work with what I did, because I am returning a pointer to uint8_t, which I am not sure if that is what they wanted or if I should have cast it into a char. At any rate hopefully this will work for short patch.
 Thanks,
 Donald

sodcorec, 2016-11-01:
Sorry Short Term

Aragantinis, 2016-11-01:
As a short term solution I downloaded the code from a previous commit. But I think your solution is better.

andrewrapp, 2016-11-02:
change reverted

closed 2016-11-02 by andrewrapp

sodcorec, 2016-11-05:
I did not write this to the actual repository on the GitHub, I just meant I changed the file in my own computer.
 What does change reverted mean?