andrewrapp/xbee-arduino #36: At Remote command using API mode

Auron691, 2016-10-31:
Hi all, i am learning Arduino and Xbee device. I have a coordinator in API mode ( XB24-ZB ) connect to my PC using a USB board and a Router in API mode with Arduino, XBee shield and Xbee.
Both the Xbee are series 2.
I try to send a simple transmit packet from router to coordinator and it receive that, so now i want to send a AT command from router to coordinator but the coordinator dont receive the packet.
There is my code:
```
#include <XBee.h>

uint8_t irCmd[] = {'I','D'};
uint8_t irValue[] = {};
XBee xbee = XBee();

XBeeAddress64 addr64 = XBeeAddress64(0x0013A200, 0x40FC875B);
RemoteAtCommandRequest remoteAtRequest = RemoteAtCommandRequest(addr64, irCmd, irValue, sizeof(irValue));

void setup() {
Serial.begin(9600);
xbee.setSerial(Serial);
}

void loop() {
 xbee.send(remoteAtRequest);
 delay(5000);
}
```
For now i dont read the coordinator response, only send this packet. Where is my error? thanks a lot

matthijskooijman, 2017-01-19:
You might want to use sendAndWait() to see what, if any, status code is returned by the XBee.

Looking at the code, I'm wondering if `sizeof(irValue)` will return 0 as I think you expect. It could be that even empty arrays still take up at least 1 byte. So you might want to just do `RemoteAtCommandRequest(addr64, irCmd, NULL, 0)` (IIRC passing `NULL` is ok if you pass a 0 length).

matthijskooijman, 2017-01-19:
Also, how do you know the coordinator does not receive the packet? I think remote AT commands are completely handled by the remote XBee, I'm not sure that you get any indication of such a command on the remote XBee's serial port?