andrewrapp/xbee-arduino #41: Can not send data very fast?

zixzixzix, 2016-12-13:
I use
`xbee.send(zbTx);`
To send data.If i send data and wait more than 1 second every thing work fine.
But how can i send data very fast?(less than 100ms)
```
//This work fine
void loop(){
xbee.send(zbTx);
delay(1000);
}
```
//This didn't work
void loop(){
xbee.send(zbTx);
delay(100);
}

paolochiarlone, 2016-12-13:
Well that gets tricky.
Depends on how much data are you sending and how many hops.
Remember, depending on the model of the XBee, the bandwidth can be anywhere between 20 and ~100Kbps... i.e. you cannot send 100 bytes 10 times per seconds to 10 nodes with 2 hops... it is just too much!
My 2 cents.

zixzixzix, 2016-12-14:
Thank you.Bug I just send two bytes to address 0x0000 0x0000(broadcast).
I just have 3 xbee.One coordinator and two routes.
But my network may not very fine.Is it the problem?

paolochiarlone, 2016-12-14:
Well, first of all, sending data with Broadcast is the least efficient method.
If I am correct, every node can repeat the message up to 8 times.
Which means, you sent 24 messages around (8x3), not just one...
However, that should still work.
Unless, you have a bad signal, which means even more retries...
Are you using AT mode or API mode?
That is also important to know.
I would STRONGLY suggest to go through the manual of the XBee modules to understand better how the protocol and transmission works, that will help you:
http://www.digi.com/resources/documentation/DigiDocs/90002002/default.htm
Good luck!

closed 2017-08-18 by andrewrapp