andrewrapp/xbee-arduino #43: Xbee Pro 900HP support?

jpwhitney, 2017-03-22:
Does this project support the Xbee Pro 900HP modules?  I'm having some challenges using this library to send using API mode.

https://www.digi.com/pdf/ds_xbeepro900hp.pdf

kikolobo, 2018-04-06:
Hi @jpwhitney.
Had you had any luck using it?

jpwhitney, 2018-04-06:
No, it seems like it should work, but I haven't had a chance to look into it in depth.

This library is still more low level than would be ideal for my purposes. It'd be great to have a function where I could send a string from one Arduino to another by passing in the string and the 2nd Arduino's address and then have a confirmation that the signal was received successfully and accurately. But unfortunately such a function doesn't seem to exist in this library and I do not currently have the knowledge to create one.

Are you trying to do a similar thing?

HavocRC, 2019-04-18:
I'm in the same boat I've had no luck getting this to work with the 900HP