andrewrapp/xbee-arduino #46: Arduino + XBee S2B ZB : Failed when sending to a specific XBee address

TiksTriks, 2017-05-14:
Hello all,

My first words are to the amazing Andrew's work with the libraries to facilitate the use of XBee's.
I'm a newbie using XBee's and this library is an excellent tool! Thank you Andrew!

Now, my issue is, when using Arduino, I can't address to a specific XBee address.
When I try to send a packet to a specific XBee address it replies with "**Failed to send packet. Status: 0xFF**"

I'm using the "HelloSend.ino" sample code provided with the book _Building Wireless Sensor Networks Using Arduino_ by Matthijs Kooijman. Thank you Matthijs.

(note: I'm using .txt extension files to be easier to post here)

Here is my code (I have changed only the addressing) to send a packet to the XCTU XBee Router address 0x0013A2004146B59A .

```
void sendPacket() {
    // Prepare the Zigbee Transmit Request API packet
    ZBTxRequest txRequest;

    txRequest.setAddress64(0x0013A2004146B59A);  //send packet to XCTU XBee Router

    uint8_t payload[] = {'X', 'e', 'l', 'l', 'o', ',', ' ', 'w', 'o', 'r', 'l', 'd', '!'};
    txRequest.setPayload(payload, sizeof(payload));

    // And send it
    uint8_t status = xbee.sendAndWait(txRequest, 5000);
    if (status == 0) {
      DebugSerial.println(F("Succesfully sent packet"));
    } else {
      DebugSerial.print(F("Failed to send packet. Status: 0x"));
      DebugSerial.println(status, HEX);
    }
}
```

See full code here : [HelloSend.txt](https://github.com/andrewrapp/xbee-arduino/files/999481/HelloSend.txt)

I also tried a different approach setting the address using XBeeAddress64:
`XBeeAddress64 remoteAddress = XBeeAddress64(0x0013A200, 0x4146B59A); `

Also gives me an error when assigning to a specific XBee address using Arduino!
Could anyone help me on what I'm doing wrong?

My best regards!

**This is my scenario:**
- 3 XBees S2B
- Firmware (send attached profile files):

**ZigBee Coordinator API**  (connected to Arduino Uno):  0x0013A2004146B5F2
[profile_23A7_Coordinator_0x0013A2004146B5F2.txt](https://github.com/andrewrapp/xbee-arduino/files/999478/profile_23A7_Coordinator_0x0013A2004146B5F2.txt)

**ZigBee Router API** (connected to Arduino Uno):  0x0013A2004146B626
[profile_23A7_Router_0x0013A2004146B626.txt](https://github.com/andrewrapp/xbee-arduino/files/999479/profile_23A7_Router_0x0013A2004146B626.txt)

**ZigBee Router API** (connected to XCTU):  0x0013A2004146B59A
[profile_23A7_Router_0x0013A2004146B59A.txt](https://github.com/andrewrapp/xbee-arduino/files/999477/profile_23A7_Router_0x0013A2004146B59A.txt)

My Tests:
---------
**With XCTU XBee (ZigBee Router API ):**
- All XBee's are visible between each other.
- sending message from XCTU XBee to all network XBee's (address 0x000000000000FFFF) , packets are received on all network XBee's (Arduino XBee Coordinator and XCTU XBee Router).
- sending message from XCTU XBee Router to the Arduino Xbee Coordinator (address 0x0000000000000000) packet is received on Arduino XBee Coordinator.
All works fine when addressing to 0x000000000000FFFF and 0x0000000000000000 using XCTU XBee Router!

**With Arduino Xbee (ZigBee Router API ):**
- sending message from Arduino XBee Router to all network XBee's (address 0x000000000000FFFF) packets are received on all network XBee's (Arduino XBee Coordinator and XCTU XBee Router).
- sending message from Arduino XBee Router to Arduino Xbee Coordinator (address 0x0000000000000000) , packets are received on Arduino XBee Coordinator.
All works fine when addressing to 0x000000000000FFFF and 0x0000000000000000 using Arduino Xbee Router.

**With Arduino Xbee (ZigBee Coordinator API ):**
- sending message from Arduino XBee Coordinator to all network XBee's (address 0x000000000000FFFF) packets are received on all network XBee's (Arduino XBee Router and XCTU XBee Router).

All works fine when addressing to 0x000000000000FFFF using Arduino Xbee Coordinator.

andrewrapp, 2017-05-19:
You should use AP=2 however I'm not convinced that explains the issue. The FF response indicates no reply from the local xbee, as if it wasn't connected. Have you looked at the docs to see if there are any changes to the api for S2B? I don't have any S2B radios so I'm unable to troubleshoot

ibrahimuslu, 2017-08-03:
Hello,
As i struggling issue like that, i have realized and fixed following code in the library 
`void XBee::sendByte(uint8_t b, bool escape) {

	if (escape && (b == START_BYTE || b == ESCAPE || b == XON || b == XOFF)) {
		write(ESCAPE);
		write(b ^ 0x20);
	} else {
		write(b);
	}
}`
as this : 
`void XBee::sendByte(uint8_t b, bool escape) {

	//if (escape && (b == START_BYTE || b == ESCAPE || b == XON || b == XOFF)) {
	//	write(ESCAPE);
	//	write(b ^ 0x20);
	//} else {
		write(b);
	//}
}`

since the txZb packet fully completed. it should be send as is. But this code snippet is sending escape and 0x20 thats incomprehensible. I dont know why that code is there.

when i changed it works charmingly.

ibrahimuslu, 2017-08-03:
I have read the past issues and saw that we should configure modules as API 2 👍 
https://github.com/andrewrapp/xbee-arduino#configuration

that code snippet is for that

hamesto, 2018-01-17:
I had the same problem as TiksTriks above. But since I am using the XBee 802.15.4 modules, I fixed the problem by using Tx64Request object instead. My code is below:

hamesto, 2018-01-17:
I had the same problem as TiksTriks above. But I noticed since I am using the XBee 802.15.4 modules, I needed to use a Tx64Request object instead. Here is my code:

`#include <XBee.h>
#include <AltSoftSerial.h>

XBeeWithCallbacks xbee;

AltSoftSerial SoftSerial;
#define DebugSerial Serial
#define XBeeSerial SoftSerial
uint8_t payload[] = { 0, 0 };

void setup() {
  // Setup debug serial output
  DebugSerial.begin(115200);
  DebugSerial.println(F("Starting..."));

  // Setup XBee serial communication
  XBeeSerial.begin(9600);
  xbee.begin(XBeeSerial);
  delay(1);

  // Send a first packet right away
  sendPacket();
}

void sendPacket() {
    // Prepare the Zigbee Transmit Request API packet
    XBeeAddress64 addr64 = XBeeAddress64(0x0013A200, 0x412822B1);
    Tx64Request txRequest = Tx64Request(addr64, payload, sizeof(payload));

    uint8_t payload[] = {'H', 'e', 'l', 'l', 'o', ',', ' ', 'w', 'o', 'r', 'l', 'd', '!'};
    txRequest.setPayload(payload, sizeof(payload));

    // And send it
    uint8_t status = xbee.sendAndWait(txRequest, 5000);
    if (status == 0) {
      DebugSerial.println(F("Succesfully sent packet"));
    } else {
      DebugSerial.print(F("Failed to send packet. Status: 0x"));
      DebugSerial.println(status, HEX);
    }
}

unsigned long last_tx_time = 0;

void loop() {
  // Check the serial port to see if there is a new packet available
  xbee.loop();

  // Send a packet every 10 seconds
  if (millis() - last_tx_time > 10000) {
    last_tx_time = millis();
    sendPacket();
  }
}`

But now I believe I am not receiving the code correctly at the receiving end. Here is my code at the receiving end:

`#include <XBee.h>
#include <Printers.h>
#include <AltSoftSerial.h>

XBeeWithCallbacks xbee;

AltSoftSerial SoftSerial;
#define DebugSerial Serial
#define XBeeSerial SoftSerial

void setup() {
  // Setup debug serial output
  DebugSerial.begin(115200);
  DebugSerial.println(F("Starting..."));

  // Setup XBee serial communication
  XBeeSerial.begin(9600);
  xbee.begin(XBeeSerial);
  delay(1);

  // Setup callbacks
  xbee.onZBRxResponse(processRxPacket);
}

void processRxPacket(ZBRxResponse& rx, uintptr_t) {
  DebugSerial.print(F("Received packet from "));
  printHex(DebugSerial, rx.getRemoteAddress64());
  DebugSerial.println();
  DebugSerial.print(F("Payload: "));
  DebugSerial.write(rx.getData(), rx.getDataLength());
  DebugSerial.println();
}

void loop() {
  // Check the serial port to see if there is a new packet available
  xbee.loop();
}`

If anyone has any suggestions or solution to fixing my receiving end, that would be greatly appreciated. Thanks!

jialarcon, 2018-10-11:
Hi, I am having the exact same problem with XBee Series 2, always getting the timeout code 0xFF if I try to address an specific router...
TiksTriks, did you finally solved the problem ?
Regards

jialarcon, 2018-10-12:
Hey ! I just double checked to see which API conf I had and it was wrong !, After switching to API 2 (escaped) it works perfectly so I can use full specific addressing.
Regards