andrewrapp/xbee-arduino #59: Zdp Scan function is not searching unless serial monitor is ON

jyotirajsharma, 2018-02-26:
Hi,
I am using Arduino UNO R3.
It seems device discovery/LQI inquiry function is not working unless serial monitor is opened.
It works only when serial monitor is ON.  Can we get the discovery function working without serial monitor on (i.e. without connected to PC, only powered on using battery)?

Please REPLY as soon as possible. 

My setup() function is given as below.

```
void setup() {
  Serial.begin(9600);
  //xbee.setSerial(Serial);

  // Set AO=1 to receive explicit RX frames
  // Because this does not write to flash with WR, AO should be reverted
  // on reboot.
  uint8_t value = 1;
  AtCommandRequest req((uint8_t*)"AO", &value, sizeof(value));
  req.setFrameId(xbee.getNextFrameId());
  uint8_t status = xbee.sendAndWait(req, 150);
  if (status == 0)
    Serial.println(F("Set AO=1"));
  else
    Serial.println(F("Failed to set AO, expect problems"));
  scan_network();
}
```

Thanks & Regards,
Jyoti Raj Sharma

matthijskooijman, 2018-02-27:
I do not see anything in the sketch that would cause it to only work when the serial monitor is on. In fact, the main MCU on the Uno doesn't even *know* when the serial monitor is open or closed.

I'm also a bit confused about your hardware setup: It seems you're using `Serial` both for debug output to the computer, as well as to talk to the XBee? That will probably be problematic, though I'm not entirely sure why it would be different based on whether the serial port is opened or not...

jyotirajsharma, 2018-02-27:
Can you confirm that if your Zdpscan.ino works well with single serial device such as Arduino Uno R3 as well ?

If I comment `Serial.begin(9600)` from `setup()` function, I can NOT see any logs in the serial monitor. But this does NOT make the scan function working. I confirmed checking `nodes_found` flag. 

Please suggest how to fix this problem.

```
setup()
{
//Serial.begin(9600);
xbee.setSerial(Serial);
xbee. begin(Serial);
}

```

matthijskooijman, 2018-02-27:
The ZdpScan.ino example was written for devices with two serial ports, since you cannot use a single serial port to talk to both the XBee and the PC.

> If I comment Serial.begin(9600) from setup() function, I can NOT see any logs in the serial monitor. But this does NOT make the scan function working. I confirmed checking nodes_found variable.

Obviously: If you do not enable the serial port, you cannot use it for anything (not for logging to the PC, not for talking to the XBee.

I would suggest you look at SoftwareSerial or AltSoftSerial for providing a second port to talk to the XBee module.

Also, I get the feeling that you don't really know what you're doing, and are mostly using and changing examples without really understanding what they do or what you're changing. I suggest you dive in deeper to better understand the code you are working with, which is the only way to really get productive in my experience. Also, I don't have time for extensive hand-holding, so if I don't reply here anymore, that's why.

jyotirajsharma, 2018-02-27:
But, I am still NOT able to figure it out why following code does NOT work, without serial monitor open (for a single serial port device)? I think  this needs a fix.
`void setup() {
  Serial.begin(9600);
  xbee.setSerial(Serial);
  // Set AO=1 to receive explicit RX frames
  // Because this does not write to flash with WR, AO should be reverted
  // on reboot.
  uint8_t value = 1;
  AtCommandRequest req((uint8_t*)"AO", &value, sizeof(value));
  req.setFrameId(xbee.getNextFrameId());
  uint8_t status = xbee.sendAndWait(req, 150);
  if (status == 0)
    Serial.println(F("Set AO=1"));
  else
    Serial.println(F("Failed to set AO, expect problems"));
  scan_network();
}`

jyotirajsharma, 2018-02-27:
Thank you for your suggestion. It seems now it is working with using Software Serial.
```
SoftwareSerial XBee_serial(2, 3); // RX, TX
 XBee_serial.begin(9600);
  Serial.begin(9600);
  xbee.setSerial(XBee_serial);
```