andrewrapp/xbee-arduino #65: Ack packet from receiver

victors0991, 2018-05-19:
I have two xbee s2b one coordinator and one router. I want to know if there is a way that receiver send ack when receiver recieve data from transmitter. ?

victors0991, 2018-05-20:
Yes I tried and its work. I have a last question. I have a problem now because xbee is booting faster that arduino so when xbee transmiter send data to xbee receiver, so xbee receiver replay with ack but that data never go to arduino because it is booting so I put power of xbee into digital pin in arduino so when arduino is booted so send power to that pin and xbee power and but that is not work because xbee is power but with a little voltaje becausd xbee's leds are power up but not brigther than when I send voltaje to xbee so how do can I do in order to power up xbee when arduino is booted. My xbee is in router mode

JChristensen, 2018-05-20:
If only a small amount of data is being sent, the XBee should hold it in a buffer until the Arduino reads it.

It's a bad idea trying to power the XBee from an Arduino pin, it is not capable of supplying sufficient current to properly power the XBee. This could potentially damage the microcontroller. Also beware if the Arduino operates on 5V because the XBee is a 3.3V device and too much voltage will damage it.

I might use the XBee's reset pin. Hold it in reset until the Arduino is ready to talk to it. If the Arduino operates at 5V, a small level-shifting circuit will be needed to connect to the XBee reset pin.

victors0991, 2018-05-20:
Yes, me too. But if I use reset pin of xbee, I will erase all configuration and that I dont want to use. Is there any way to make xbee wait until arduino is booted?

JChristensen, 2018-05-20:
As long as configuration changes have been written to non-volatile memory, a reset will not erase them.

victors0991, 2018-05-20:
I will try. I read the xbee-arduino library but what means xbeecallback function? If it uses when we want something to return?