andrewrapp/xbee-arduino #75: Dose this library work with XBee S2C？

408881465, 2019-03-16:
Dose this library work with XBee S2C？

JChristensen, 2019-03-16:
Yes, I have been using it with S2C modules with no issues.

matDeng, 2019-04-15:
Does this library work if i use xbee shield that attach with arduino ??

dheijl, 2019-07-06:
Yes, I used S2C modules with an Arduino Uno (using softwareserial on pins 2 and 3) and with an Adafruit Metro M4 Express (using Serial1 on pins 0 and 1, a hardware UART).

ChetanAmrao, 2019-07-07:
how do i send zbStatusResponse from a Router using send() to the Coordinator?
I cannot use ZBTxRequest for it, can I? i might be wrong 
I used the Series2_Tx on the Coordinator and it doesn't receive any StatusResponse so i have to resort to using ZB_EXPLICIT_RX_RESPONSE which i want to avoid now

any guidance would really help me thanks
cheers!

claudioandres1677, 2019-09-08:
CONECTAR EL ARDUINO DUE Y XBEE 2 O  S2C CON ESTÁ LIBRERÍA NO TENDRÉ PROBLEMAS ?

claudioandres1677, 2019-09-08:
CONNECT THE ARDUINO DUE AND XBEE 2 OR S2C WITH THIS LIBRARY , WILL NOT HAVE PROBLEMS?