andrewrapp/xbee-arduino #88: Doesn't work with XBee 3

SWoto, 2020-05-12:
Hello,

I am trying to get the XBee3 working with Arduino and another one connected to XCTU, however, the code that I used for XBee2 didn't work for the Series 3.

Does anyone how to correct it?

AustinErck, 2020-05-15:
I've been having an issue in the same realm. I have a [series 3](https://www.sparkfun.com/products/15126) Xbee, and when I run the same script it always gets to the "Not timely TX status response" else statement. However they seem to work as in XCTU they create a network and communicate the FN AT command. I'm using API mode, obviously, with Digimesh

rm-rf42, 2020-10-22:
Any one have an update to this? Trying from an Arduino IOT 33 using the TX/RX pins on it (Serial1).  I am using a Things Plus xbee series 3 though.

@AustinErck  I am using the same setup (API/Diigimesh).

I have tested with a separate TX/RX Serial cable that both RX/TX ends of the Arduino and xbee series 3 things plus work perfect. I have had other issues though when simply trying to send Arduino(TX) to xbee3(RX) simple UART traffic (single characters).

mochiecake, 2020-12-08:
I am experiencing the same issue. Unable to send a single character over to XBee3. I have tried this example: http://www.science.smith.edu/dftwiki/index.php/Tutorial:_Arduino_and_XBee_Communication , but not getting anything from XCTU console.

keeper2597, 2021-03-01:
I'm fairly new to this stuff but if it's helpful to anyone else - I'm using Series3 XBees with this library and it largely works IF you're using the right protocols.  I think it's clearer to say that what is defined as SERIES_1 here is actually better thought of as 802.15.4 protocol.  And rather than SERIES_2 it's actually Zigbee ZH.  So if you use the examples, XBees set to 802.15.4 Firmware should use Series1 and XBees set to Zigbee ZH should use Series2.  I think this library would need some updating to work with the DigiMesh protocols.

ledbutter, 2021-08-30:
Not sure if this will solve the problem for anyone else, but I could not get Tx requests to go through until I updated API mode (in XCTU) to be "API Mode with Escapes [2]". I finally thought to try this after going over the source code comments again and saw [this](https://github.com/andrewrapp/xbee-arduino/blob/39f236ea1827fd5fe3ce66ffcc07433ab8ab9197/XBee.h#L744):
```
// The XBee radio must be configured in API (packet) mode (AP=2)
```
I was seeing the same behavior as everyone else with it in "API Mode without Escapes [1]" and did not know enough to know whether it mattered or not. Clearly, in my case it did.

PaisBruno, 2021-12-14:
> Not sure if this will solve the problem for anyone else, but I could not get Tx requests to go through until I updated API mode (in XCTU) to be "API Mode with Escapes [2]". I finally thought to try [...]

Worked for me! Thanks ledbutter.

myherdemGumbel, 2024-04-17:
For those who are struggling like me after upgrading from xbee2 to xbee3. 
Digimesh topology didn't work with this lirary. So I updated the firmware to Zigbee.
Also I used Api mode as AP=2 as ledbutter mentioned.