andrewrapp/xbee-arduino #91: Software Fix

Trajectory989, 2021-01-23:
Hi guys. I think it may help to solve some problems if we can update the XBee::send(XBeeRequest &request) code to account for escaped characters. 

basically if a byte is 0x13 (which is in many MAC Address for XBees -- hence the 64bit addressing issues people have been having), 0x7E, 0x7D, or 0x11, a 0x7D byte should be added in before it and that byte XORed with 0x20. This should not affect the length or the checksum of the message, so perhaps a byte-by-byte loop through the message before calling flush() in this function could easily fix the issue many are having.

For more documentation see here: https://www.digi.com/resources/documentation/Digidocs/90001942-13/concepts/c_api_escaped_operating_mode.htm?TocPath=XBee%20API%20mode%7COperating%20mode%20configuration%7C_____1

The current code just needs updated to work with escaping, which I believe could be an easy fix.

matthijskooijman, 2021-01-23:
Hm, it has been quite a while since I worked with this code, but from what I remember, this escaping is already implemented in this library. On the device side, it is enabled by setting `AP=2` (API mode with escaping), as opposed to `AP=1` (API mode without escaping) or `AP=0` (AT mode). See also https://github.com/andrewrapp/xbee-api/wiki/WhyApiMode

Are you *sure* that this is something that needs to change on the library side? Or did you maybe set `AP=1` and have problems because of that?

closed 2021-01-23 by Trajectory989

Trajectory989, 2021-01-23:
Well I will admit I am able to make the same frame as XCTU, but I still have the issue where I am not sending properly from the Arduino side. If I send something to the Arduino using XTCU, the Arduino recognizes that it received something and gets the payload fine. The Computer-connected Xbee gets its transmit status and all is well in the world one way (from computer to Arduino). The issue is the other way around....I can testify that the frame is correct, but it is not received.